pointhi/kicad-footprint-generator #19: KiCad Use

SchrodingersGat, 2017-01-18:
Hi @pointhi 

Would you mind if I made a copy of this repo on the /kicad/ github page? That way we can provide an officially maintained set of footprint scripts?

Cheers

pointhi, 2017-01-19:
For reference: https://github.com/KiCad/kicad-library-utils/pull/78

I think It would be good idea to place those scripts into an official repository.

My request would be to let KicadModTree stay as clean and tested as possible (means clean implementations of new features and no hacks if possible). Probably as git subproject, to have a bit of separation from the scripts.

SchrodingersGat, 2017-01-19:
@pointhi for clarification:

Would you like the kicad-library-utils to *link* to KicadModTree (on your repository)? Or are you ok with us  making a copy of the files?

pointhi, 2017-01-20:
Well, making a copy of the framework is ok with me.

SchrodingersGat, 2017-01-21:
Ok, thanks @pointhi 

It is a fantastic tool that you have made :) such a huge amount of the kicad library is thanks to you!

closed 2017-01-28 by pointhi