pointhi/kicad-footprint-generator #38: Exception in the example

matyro, 2017-04-29:
Hi,
i get following exception when i execute the example from the webpage or in the example directory.

I am using anaconda 4.3 (python 3)

Traceback (most recent call last):
  File "./simpleFootprint.py", line 39, in <module>
    kicad_mod.append(RectLine(start=[-2, -2], end=[5, 2], layer='F.SilkS'))
  File "/home/username/workspace/kicad-footprint-generator/KicadModTree/examples/../../KicadModTree/nodes/specialized/RectLine.py", line 84, in __init__
    PolygoneLine.__init__(self, polygone=polygone_line, layer=kwargs['layer'], width=kwargs['width'])
KeyError: 'width'

Is the problem on my end?

matyro, 2017-05-04:
Problem was on my end and is fixed now

closed 2017-05-04 by matyro

pointhi, 2017-09-13:
That was a bug which is now fixed