pointhi/kicad-footprint-generator #64: Script for "standard" male shrouded headers

evanshultz, 2017-10-06:
There are loads of headers with this general description, dual row in my experience. It seems like a standard part and not manufacturer-specific, since loads of vendors make them.

For clarity about what I'm describing, here is a picture:
[image]

Are there footprints for any of these variants in the library? I confess that I don't see it, but I'm so surprised I feel I must be missing it. https://github.com/KiCad/Socket_Strips.pretty has something similar, but no shrouds or latches (just bare breakaway strips). If there are footprints and/or a script, I'd be obliged if someone can point me to it so I can cover all the common varieties. If there's nothing already, I'll start from scratch.

Here is my wish list (not every single combination exists in the market and would be created):
- Mounting: SMT, THT
- Pitch: 1.27mm, 2.00mm, 2.54mm
- Rows: 2
- Pin count (even only): 4-44, 50, 60, 64, 68, 80, 100, 110, 120
- Orientation: Straight, Angled
- Latch: With, without (default and unannotated in footprint name)

One last note: My gut is to generate them all from the script for simplicity and maintainability. Would that be an issue?

@pointhi @SchrodingersGat @poeschlr @jkriege2 @hackscribble

pointhi, 2017-10-06:
I created such footprints before: https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Multicomp

we can improve the script to use the new argparse interface, and then we would only need to create csv files specifying the different variations. (Which could be generated)

evanshultz, 2017-10-06:
Thanks @pointhi!

Yes, this is similar to what I was thinking. Albeit older and using kicad_mod. And not meeting the current KLC. And without latches. And THT only. And using a separate launcher shell script. Hmmm...

evanshultz, 2017-10-07:
@pointhi @SchrodingersGat 
Why is the argparse interface used and encouraged? It allows sending arguments, such as the Molex 52991 script, to generate the different footprints instead of having it all self-contained where executing the script generates all footprints directly.

I can see that having argparse allows new footprints to be created simply by running the script with new arguments. The script doesn't need to be touched. This is handy if a series of parts gets updated to add new varieties.

On the other hand, it means an external script or manual work (defeating the purpose of automation) is needed to generate all footprints.

What was the reason? I'm just asking since I see good and bad on both sides.

Perhaps the script could support parameters, but if none are provided it will use an internal hard-coded list to generate the footprints? That way the script can be written to generate the whole family of parts automatically, but if the user wants to supply arguments the script will also build footprints in a one-off fashion? Sound good?

Relatedly, looking at the Molex 52991 script it doesn't guard against improper pin counts (odd counts, no bounds on range, etc.) nor improper Molex PN suffix (must only be 40 digits, etc.). In fact, the suffix could be used to derive the pin count by stripping off the first 2 digits. These seem like nice enhancements to me; any thoughts or pushback against that?

SchrodingersGat, 2017-10-07:
I think it is a matter of preference. My scripts always included the model data hard-coded into the .py file itself, so that's always an option.

pointhi, 2017-10-08:
Well, I see various points why the argparse interface should be used:

* it gives users a general interface which is used for all scripts
* it allows users to modify parameters without diving into the source code (at the moment, everyone has it's own schema to store data, including problems like position depending arguments (which are not self describing)
* for this part for example, many variants exists (from various vendors). Having a definition file for every series is way cleaner than writing python scripts for them or extending existing ones
* those files can be used in other tools (like 3d-model-generators) as well
* it helps non-programmer to use existing scripts
* when we have a general interface, we can build tools like a graphical footprint-wizard later without changing the scripts

the point with the slightly more comlicated execution is true. But that's only adding one argument.
We could for example use Makefiles in the future to hardcode such steps, so you could for example run ```make molex_straight```. This would also allow us to build all libraries with a single command, to adjust them for other courtjard distances for example (When we implement overrides into the parser). 

Look into https://github.com/pointhi/kicad-footprint-generator/issues/55 for more discussion about that

evanshultz, 2017-10-09:
To finish off PRs which never got completed and would be ideal for scripting, I'm working on https://github.com/KiCad/Connectors_Molex.pretty/pull/35. Molex 502430 is the first one I started on.

Using the 72991 script as a seed (https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connectors_Molex/slimstack-52991.py), I developed the following script which accepts one optional parameter for the PN suffix. If no argument is given, all footprints currently  listed in the Molex datasheets are generated.

A few other changes which aren't specific to the part:
- Only one parameter is used (pincount is derived from the PN suffix)
- PN suffixes are validated (this could be generalized to accept only 4 digits with an even number for the first two digits, only 1 or 2 or 3 for the third digit, and a 0 as the last digit).

Could I get some feedback on this implementation? Edge cases I haven't considered? Improvements?

Here is the script
```
#!/usr/bin/env python

# KicadModTree is free software: you can redistribute it and/or
# modify it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
#
# KicadModTree is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with kicad-footprint-generator. If not, see < http://www.gnu.org/licenses/ >.
#
# (C) 2016 by Thomas Pointhuber, <thomas.pointhuber@gmx.at>

import sys
import os
import argparse

sys.path.append(os.path.join(sys.path[0], "../.."))  # enable package import from parent directory

from KicadModTree import *  # NOQA

def round_to(n, precision):
    correction = 0.5 if n >= 0 else -0.5
    return int( n/precision+correction ) * precision

if __name__ == '__main__':

    # handle arguments
    parser = argparse.ArgumentParser()
    #parser.add_argument('pincount', help='number of pins of the connector', type=int, nargs=1)
    parser.add_argument('gen_pn', help='suffix to 502430 series number (e.g. 0820)', type=str, nargs='*')
    parser.add_argument('-v', '--verbose', help='show extra information while generating the footprint', action='store_true')
    args = parser.parse_args()
    gen_pn = args.gen_pn

    # list of valid partnumber suffixes in all family datasheets
    valid_pns = ["1410","2010","2210","2410","2610","3010","3210","3410","4010","4410","5010","6010","6410","8010","0820","1430","2030","2230","2430","2630","3230","4030","5030","6030","7030","8030","9030"]
    partnumbers = valid_pns

    if len(gen_pn) == 1:
        partnumbers = gen_pn
    
    for partnumber in partnumbers:
        
        # do not proceed if pin count or partnumber are not valid
        if partnumber not in valid_pns:
            sys.exit("Partnumber is not valid!")
        
        # the first two digits of the PN suffix are the pin count
        pincount = int(partnumber[:2])
        
        footprint_name = 'Molex_SlimStack_Receptacle_2x{pc:02g}_Pitch0.5mm_502430-{pn:s}'.format(pc=pincount/2, pn=partnumber)
        
        print('Building {:s} '.format(footprint_name))
[... 61 lines cut ...]
        kicad_mod.append(RectLine(start=[-h_body_length, -h_body_width], end=[h_body_length, h_body_width], layer='F.Fab', width=fab_width))
        kicad_mod.append(Line(start=[-h_body_length+outline_x, -h_body_width], end=[-h_body_length+outline_x, -h_body_width-marker_y], layer='F.Fab', width=fab_width))

        # create silkscreen outline and pin 1 marker
        left_outline = [[-h_body_length+outline_x, h_body_width+nudge], [-h_body_length-nudge, h_body_width+nudge], [-h_body_length-nudge, -h_body_width-nudge],\
                        [-h_body_length+outline_x, -h_body_width-nudge], [-h_body_length+outline_x, -h_body_width-marker_y]]
        right_outline = [[h_body_length-outline_x, h_body_width+nudge], [h_body_length+nudge, h_body_width+nudge], [h_body_length+nudge, -h_body_width-nudge],\
                         [h_body_length-outline_x, -h_body_width-nudge]]
        kicad_mod.append(PolygoneLine(polygone=left_outline, layer='F.SilkS', width=silk_width))
        kicad_mod.append(PolygoneLine(polygone=right_outline, layer='F.SilkS', width=silk_width))

        # create courtyard
        kicad_mod.append(RectLine(start=[-courtyard_x, -courtyard_y], end=[courtyard_x, courtyard_y], layer='F.CrtYd', width=courtyard_width))

        # add model
        kicad_mod.append(Model(filename="${{KISYS3DMOD}}/Connectors_Molex.3dshapes/{:s}.wrl".format(footprint_name), at=[0, 0, 0], scale=[1, 1, 1], rotate=[0, 0, 0]))

        # print render tree
        if args.verbose:
            print(kicad_mod.getRenderTree())

        # write file
        file_handler = KicadFileHandler(kicad_mod)
        file_handler.writeFile('{:s}.kicad_mod'.format(footprint_name))
```

evanshultz, 2017-10-11:
@pointhi @SchrodingersGat 
Would you mind taking a look at the script above and seeing if that works for you? How can it be improved before I start using this method to submit footprints and scripts?

evanshultz, 2017-10-11:
@pointhi @SchrodingersGat Thoughts or comments on my script above?

SchrodingersGat, 2017-10-12:
Looks sensible to me :)

evanshultz, 2017-10-16:
Here is a simplified and improved version of the above (only the section below has changed). I believe all requirements are met, in that all current PNs are hard-coded and can be generated at once but an argument is also allowed to generate PNs one-by-one. I'm satisfied now and will leave this Issue open for a few days to gather feedback before submitting the script.

```
if __name__ == '__main__':

    # handle arguments
    parser = argparse.ArgumentParser()
    #parser.add_argument('pincount', help='number of pins of the connector', type=int, nargs=1)
    parser.add_argument('gen_pn', help='suffix of 502430 series number (e.g. 0820)', type=str, nargs='*')
    parser.add_argument('-v', '--verbose', help='show extra information while generating the footprint', action='store_true')
    args = parser.parse_args()
    gen_pn = args.gen_pn # input argument in a list

    # list of valid partnumber suffixes from all datasheets
    valid_pns = ["1410","2010","2210","2410","2610","3010","3210","3410","4010","4410","5010","6010","6410","8010","0820","1430","2030","2230","2430","2630","3230","4030","5030","6030","7030","8030","9030"]
    
    # if user did not supply an argument we will get an empty list and generate all footprints (PN validation is later)
    if gen_pn:
        partnumbers = gen_pn
    else:
        partnumbers = valid_pns
    
    for partnumber in partnumbers:
        
        # do not proceed if pin count or PN are not valid
        #if not ((partnumber.isdigit()) and (len(partnumber) == 4) and (int(partnumber[:2]) % 2.0 == 0) and (partnumber[2:3] in ["1","2","3"]) and (partnumber[3:4] == "0")):
        if partnumber not in valid_pns:
            sys.exit("Partnumber is not valid!")
        
        # the first two digits of the PN suffix are the pin count
        pincount = int(partnumber[:2])
        
        footprint_name = 'Molex_SlimStack_Receptacle_2x{pc:02g}_Pitch0.5mm_502430-{pn:s}'.format(pc=pincount/2, pn=partnumber)
```

evanshultz, 2017-10-17:
I think the above demonstrates the use of optional arguments nicely, along with other small improvements. However, other comments on the script are welcome.

Are there objections or suggestions to the scope of footprints I mentioned in the initial comment? If not, I'll add scripts for these footprints to my todo list.

closed 2017-10-26 by evanshultz

matthijskooijman, 2018-11-19:
@evanshultz, did you ever get around to generating the footprints listed in the original comment? I've been looking for an 1.27mm and 2.0mm shrouded pin header footprint, but I can only find 2.54mm versions in Connector_IDC and Connector_Multicomp.

evanshultz, 2018-11-19:
I started on it and made some comments at https://github.com/KiCad/kicad-footprints/pull/41, but then got side-tracked with other things. It is on my list unless somebody beats me to it, but I wouldn't expect to get to this for another month.

matthijskooijman, 2018-11-20:
Thanks for the update. I also found https://github.com/pointhi/kicad-footprint-generator/issues/134 which seems to track the progress of such a generator script for shrouded headers.