pointhi/kicad-footprint-generator #134: IDC connector footprints

evanshultz, 2018-06-18:
I have a few questions about the IDC header footprints:
1. Is there a script to generate them? An example footprint is `IDC-Header_2x13_P2.54mm_Vertical_Lock`.
1. The vertical footprints have large THT pads which aren't in the [referenced datasheet](https://www.reboul.fr/storage/00003af6.pdf).
1. The vertical footprints don't show the latch in the open position, which means any required keepout for any tall components or other elements near the PCB can't be easily found. I've seen this shown in silk on every PCB using this type of header I've ever seen.
[image]
1. The horizontal header is potentially even worse because no components can lay on the PCB (or near the edge if the connector doesn't hang totally off) and the open position of the latches aren't shown.
1. The horizontal is called `Lock` but it doesn't include latches in the drawing but only the shrouded header that can optionally support latches!
[image]
1. Latches come in various sizes, so the vertical and horizontal footprints may need a couple different footprints to cover short and long latches. Even if they vary a little bit, certainly a reasonable generic footprint could be created.
1. Using `Lock` seems like a poor word choice. Locking can be achieved with a ramp or other means for a similar connector, but latches are more specific to this type of lock. In other words, latches seem to me to be a subcategory of locks and we could be more specific in our footprint names.
1. Would it be best to reference a standard for these connectors? DIN 41651, I believe, is the standard for these connectors (but I don't have it to review).

evanshultz, 2018-07-20:
Anybody?

evanshultz, 2018-08-09:
Nobody? On any item above?

evanshultz, 2018-08-22:
@Shackmeister @terjeio 
Do either of you know anything about the footprints I've mentioned above?

matthijskooijman, 2018-11-20:
The scripts at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Multicomp might also be relevant. These generate footprints for the multicomp MC9A12 and MC9A22, which are 2.54mm vertical and horizontal shrouded pin headers. The footprints are different from the ones in the Connector_IDC library, so this is probably not the same script. I'm not sure whether the script is a good one to base upon in terms of use of the framework (but since it is three years old, it might not).

evanshultz, 2018-11-20:
Yes, I couldn't find the scripts that generated any of the existing footprints so I assume they were hand made (I didn't check the Git history).

The scripts you mention use the old framework and are not a good starting point. While they work, starting over is probably best if one of the modern scripts (using KiCadModTree).

If you're wanting to do this now and not wait for me, here are the notes I've kept:
```
box headers with latches (DIN 41651?)
https://www.3m.com/3M/en_US/company-us/all-3m-products/~/3M-Four-Wall-Header-3000-Series/?N=5002385+3293481906&preselect=8709318+8710652+8733900+8734573&rt=rud
https://multimedia.3m.com/mws/media/22268O/3mtm-100-in-4-wll-hdr-100x-100l-e-strt-cmpl-pin-ts0478.pdf (**overall dimensions**)
http://www.selecom.it/pdf/06din416.pdf (**overall dimensions**)
https://www.reboul.fr/storage/00003af6.pdf (**overall dimensions**)
http://www.oupiin.com/product_iii.html?c1=10&c2=54
http://www.oupiin.com/product_iii.html?c1=10&c2=52
http://www.assmann-wsw.com/fileadmin/catalogue/04_Multiflex_rev4-0.pdf
```

matthijskooijman, 2018-11-20:
I was looking at this now, though I might not find enough time to finish this through. Still, let me bother you with a question then. I'm looking at the existing script,  and assume that [main_generator.py](https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Conn_PinSocket/main_generator.py) is the entry point. However, it imports `parameters`, but I can't find what module this refers to: https://github.com/pointhi/kicad-footprint-generator/blob/70954af68bd1d8ad57059e029d0620296024a36c/scripts/Conn_PinSocket/main_generator.py#L34

I can find [`parameters.yml`](https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Conn_PinSocket/parameters.yaml) that I think should eventually be loaded by (the code imported by) this import, but a YAML file is not directly importable of course. Am I missing something here? Is there a tool to convert the YAML to a py file perhaps?

Note that I only tried this with python3 (fixing the `print` call in the script), so perhaps there is some py2-only magic to make this work, but I did not have a good internet connection to try py2 before (and no time right now).

evanshultz, 2018-11-20:
I ran into the exact same issues and wrote them down at https://github.com/KiCad/kicad-footprints/pull/41#issuecomment-419216638. I then abandoned this task, hoping to get a response while I finished off other tasks. I'll probably start over with a new script when time allows.

matthijskooijman, 2018-11-20:
Ok, I made the Conn_PinSocket script work, see #222.

Looking at the script, I wonder if it is the best starting point. It adds its own "canvas.py" as a layer on top of the main framework for, I think, keepout areas (see #82 for discussion about it), as well as it's own unique parameter handling with quite some boilerplate. It's not completely clear how necessary these parts are, but if they are, they should probably be integrated into the main framework rather than duplicating it for a new IDC script....

I'm out of time for now, I'll probably check back here when I next have time for the PCB that I need all this for :-)

evanshultz, 2019-02-20:
FYI (to all), I'm starting on this script now.

The Multicomp footprints in `Connector_Multicomp` are generic and should not should not have their own library (my opinion). IDC (insulation displacement contact) describes the type of technology to electrically mate the ribbon cable wires and header using forks to slice the ribbon cable insulation; so to me `Connector_IDC` as a library name seems poorly descriptive. Creating a script to re-create the four footprint variants there shouldn't be too hard, but there are other variants that are missing now. I plan to replace both those libraries and put footprints in `Connector_PinHeader_Box_P<pitch>mm` and `Connector_PinHeader_Latch_P<pitch>mm`, the latter having being like this:
[image]

I will explore basing this off of the script at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Connector_PinSocket, only because it's newer even if it's overly complicated, with a fallback to https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/pin-headers_socket-strips or https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Multicomp. The "Lock" footpritnts appear not to have a script so there will clearly be some real work no matter where I start.

Scope of footprints to make (not every single combination would be created):
- Body style: Box, Latch
- Latch Lengths (latch only): 6.5mm, 9.5mm, 12mm
- Mounting screw (latch only): Yes, No
- Pitch: 1.27mm, 2.00mm, 2.54mm
- Rows: 2
- Pin counts:
-- Box: 4-44, 50, 60, 64, 68, 80, 100, 110
-- Latched: 6, 8, 10, 14, 16, 20, 24, 26, 30, 34, 40, 44, 50, 60, 64
- Orientation: Straight, Angled

Latched header information that shows the latch length:
```
https://multimedia.3m.com/mws/media/22268O/3mtm-100-in-4-wll-hdr-100x-100l-e-strt-cmpl-pin-ts0478.pdf
https://cdn.amphenol-icc.com/media/wysiwyg/files/drawing/71918.pdf
http://www.selecom.it/pdf/06din416.pdf
https://www.reboul.fr/storage/00003af6.pdf
```

evanshultz, 2019-03-27:
I've been a bit delayed, but I finally finished putting together the [spreadsheet of info](https://docs.google.com/spreadsheets/d/16SsEcesNF15N3Lb4niX7dcUr-NY5_MFPQhobNuNppn4/edit?usp=sharing) for connector dimensions. I'm starting with vertical 2.54mm pitch.

poeschlr, 2020-05-07:
@evanshultz i think you forgot to push the latest changes (you added the spreadsheet link to the documentation of the footprints but this is missing here)

closed 2020-05-07 by evanshultz