pointhi/kicad-footprint-generator #154: Package KicadModTree

maximeborges, 2018-07-23:
It took me a few minutes to understand how to launch the scripts.
For reference, I copied the script in the root directory of the repo, and finally copied the KicadModTree/ to site-packages to be able to run any script anywhere.
I was searching for a way to properly install it, as any other module, but can't find anything.
Am I missing something or isn't there any way to correctly install this as a module at the moment ?

pointhi, 2018-07-23:
You are true, I'm missing a setup.py at the moment and calling the script is managed by some small hack (depending on the location of the script):

```python
import sys
import os

sys.path.append(os.path.join(sys.path[0], "..", "..")) # load parent path of  KicadModTree
```

I will look into it. This is a long overdue issue.

closed 2018-08-02 by pointhi

pointhi, 2018-08-02:
KicadModTree now avaliable on PyPI: https://pypi.org/project/KicadModTree/

commit cdb56958 (2018-08-02): Initial setup.py, FIX: #154