pointhi/kicad-footprint-generator #174: Recommended thickness of solder paste

jneiva08, 2018-09-10:
I'm trying to make a script to new connector.

Did successful make smd pad and then pad with F.Paste with different sizes.
In the datasheet says "Recommended thickness of solder paste: 120um"
How can add in script?

evanshultz, 2018-09-10:
You cannot control this in the footprint as it will be determined by the stencil thickness. And the stencil thickness will be determined by a great many factors that are much bigger in scope than just one footprint. (Sometimes there is one particular footprint on a PCB that forces some requirement about the stencil, but in my experience it's typically considering all parts on the PCB as a whole.)

So, you don't need to do anything with that note in the datasheet. This could be captured by a note in the footprint description, though.

jneiva08, 2018-09-10:
Thanks for fast answer

closed 2018-09-10 by jneiva08