pointhi/kicad-footprint-generator #210: Pip install fails

Munken, 2018-10-23:
Installing from PyPi fails with the following error:

```
> pip install -U KicadModTree                                                                                       
Collecting KicadModTree
  Downloading https://files.pythonhosted.org/packages/32/bf/03d74cafe091fa62cb7f96134b4ea9b32c3a328727e492fc362e263f2875/KicadModTree-1.0.3.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-Ppptc0/KicadModTree/setup.py", line 11, in <module>
        requirements = map(str.strip, open('requirements.txt').readlines())
    IOError: [Errno 2] No such file or directory: 'requirements.txt'
```

I'm running Python 2.7.15rc1 on Ubuntu 18.04.

pointhi, 2018-12-19:
@Munken still relevant? I cannot reproduce it

herostrat, 2019-05-19:
@pointhi I can reproduce this

Installation with pip3 does however work.