pointhi/kicad-footprint-generator #223: Pad script recalculate Z height into non metric distance

Misca1234, 2018-11-23:
When you create a script based foot print with the following lines

```
f = Footprint(footprint_name)
s = footprint_name.replace('Valve_', 'Valve_Tube_')
f.append(Model(filename="${{KISYS3DMOD}}/Valve.3dshapes/{}.wrl".format(s), at=(0,0,socket_H)))

```

and socket_H is 11 even according to the foot print file

```
  (model ${KISYS3DMOD}/Valve.3dshapes/Valve_Tube_Noval_B9A-9_D25.50x11.00_P11.89_A36_RP.wrl
    (at (xyz 0 0 11))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 0))
```

KiCad itself recalculate this into 279,4

[image]

Is this a known problem or should it be reported to the coders ?

Misca1234, 2018-11-25:
Also

There is a serializer function in Model.py but that is not used becaouse KicadFileHandler.py is ignoring this function and use it's own, `    def _serialize_Model(self, node):` at line 149

So, it is not possible to modify Model.py to handle both at and offset arguments to Model.py

Misca1234, 2018-11-25:
With the valve push I will include a fix both in KicadFileHandler.py and Model.py so it will handle both at and offset where at supersede offset

Misca1234, 2018-11-25:
Closing, see push
https://github.com/KiCad/kicad-footprints/pull/1111

closed 2018-11-25 by Misca1234

evanshultz, 2018-11-28:
When I create a new footprint just for testing and add a 3D model in v5 I see this:
```
  (model ${KISYS3DMOD}/Button_Switch_SMD.3dshapes/SW_DIP_x10_W6.15mm_Slide_Omron_A6H.wrl
    (at (xyz 0 0 0))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 0))
  )
```

For what it's worth, I think the wording has changed and `offset` is no longer used.

Misca1234, 2018-11-29:
I can not use 'at' for the valve socket + valve (2 3D models in one foot print)

If 'at' is used, then kicad automatically will multiply with 25.4 first time (and convert it into an offset and save it like that)

But the 'offset' is not recalculated but stays as it should be and the second 3D model is placed in the correct place