pointhi/kicad-footprint-generator #229: Which QFN script to use?

diegoherranz, 2018-12-02:
I'm just starting to dig into this repo since I wanted to create a few footprints and I like the idea of spripting it, so please forgive me if it's a silly question.

For a QFN script, which one is recommended `ipc_dfn_qfn_generator.py` or `qfn.py`?
Why are there two different ones?

Thanks!

evanshultz, 2018-12-03:
Use the IPC generator.

Just looking at `qfn.py` quickly, I suspect it was an older QFN footprint generator.

diegoherranz, 2018-12-03:
Thanks for the reply.
Actually, I may not need to create QFN footprints yet, but is it OK if I add a README to `scripts/Packages/Package_DFN_QFN` to make it clear?

Thanks!

diegoherranz, 2018-12-04:
I've submitted a PR with a readme (https://github.com/pointhi/kicad-footprint-generator/pull/232).

closed 2018-12-04 by diegoherranz