pointhi/kicad-footprint-generator #237: footprint generator for Phoenix Contact PTDA series

tolot27, 2018-12-10:
I started creating a generator (based on #236) for the PTDA series from Phoenix Contact.

[image]

If I look at /scripts/Connector/Connector_PhoenixContact/mstb.py I can see variants of the manufactur naming like "TerminalBlock_Phoenix", "Phoenix_Contact", "PhoenixContact" or even "Phoenix". Is there a rule or nomenclature for the manufacturers?

Also, the connectors exist in various pin spacings, like the Altech connectors. Shall I create a separate yaml for the different pin spacings or use arrays for parameters which differ between different pin spacings?

pointhi, 2018-12-19:
@tolot27 the scripts repo currently does not enforce much rules at the moment (which explains the different approaches and naming conventions). Try to stick to some existing type you think is sensible for now.

tolot27, 2018-12-19:
I made some progress and will create a PR soon, hopefully. I'll discuss further improvements about a common generator for connectors I thought about, after I created the PR.

tolot27, 2018-12-29:
After learning a lot about cq and cqparts, rendering of 3D shapes is nearly finished:

[image]

RoelKint, 2020-07-01:
Was this ever brought to a full functioning function?

tolot27, 2020-07-01:
It fully worked for me but I lost the script generator script. But it should be easy to regenerate them.