pointhi/kicad-footprint-generator #257: Scripting UQFN with 3 different pin widths?

EvilMav, 2019-01-04:
I want to push a part to kicad-symbols which requires the [R-PUQFN-N10](http://www.ti.com/general/docs/lit/getliterature.tsp?baseLiteratureNumber=MPQF186&fileType=pdf) package. 

As always, TI was creative with this one. If you look at the foorprint, the pins 5, 6, 7 ALL have different widths. 

Is this thing scriptable in some way, would you prefer extending the script if not or should I just go ahead and make them manually?

pointhi, 2019-01-05:
@poeschlr do you want to have the scripts extended for different pad sizes?

poeschlr, 2019-01-13:
This might be a bit much for now. But we can definitely put this on our list of things to add later.
(This would need a lot of thought on how to define the yaml file properly such that we have something generic enough but not too complex. The same would then be true for the internal data structure. The current one would not be easily able to accept such generic footprint definitions.)

EvilMav, 2019-01-17:
I will make the the footprint manually for now then.