pointhi/kicad-footprint-generator #272: BGA versions with corner markers on the copper layer?

EvilMav, 2019-01-21:
With DIY-soldering fine pitch BGAs accurate placement can get tricky: you can't see the balls from the top and the only thing you can align against is the silkscreen - if it is even used. AFAIK the tolerances of the silkscreen aren't usually great, which could be an issue aligning a 0.4mm pitch BGA. Many sources suggest putting small dots or arrows onto the copper layer at at least two opposite corners of the packages as an alignment reference. (e.g. point 7 [here](https://www.allgoodtechnology.com/bgaguide.html) )

It would probably be easy to extend the BGA script to generate a second version with a, say, "_mark" prefix which would be suitable for hand assembly by adding those marks.

pointhi, 2019-01-22:
@poeschlr opinion?

evanshultz, 2019-01-22:
The balls/bumps and circular pads will self-center when the solder paste wets, so why is such an exact placement when hand-placing a BGA required? I've not used this method and had excellent results when hand-placing and hand-soldering BGAs (with hot air, of course). And I've never seen this done on any PCB from a range of factories in a few continents.

So while this one link may request these chevrons, I would want to know if this is still current for this company and if it's widespread in the industry. My experience says that this is not necessary and it consumes potentially valuable PCB real estate.

I do agree it would be relatively simple to extend the script to add these chevrons, but I question the value of doing it and also would not be in favor of adding chevron-equipped footprints to our library.

evanshultz, 2019-01-22:
Yep, silkscreen varies a lot (compared with copper), but I've not seen it to be a problem with hand-placing BGAs. You don't need the steady hands of a brain surgeon but the self-centering is very strong so if you're careful I would expect you can get good results. You will, for sure, require soldermask; don't use BGAs without mask on the board.

poeschlr, 2019-05-29:
Sorry for the late answer. (This is kind of the wrong repo to ask such questions to be honest.)

The term you search is fiducial. These are normally not included in other footprints but are footprints in their own right. KiCad comes with some of them so feel free to use them or add your personal ones. (They are normally used to help robots align footprints but i am sure a similar concept can be used for hand placing as well.)