pointhi/kicad-footprint-generator #278: DFN packages and silkscreen - Not shown accordingly to KLC

JacobEFO, 2019-01-26:
Hey you guys.

I stumbled upon a lot of DFN packages looking like this:
[image]

In the DFN_QFN library, and I realized they were generated with the scripts here. It seems as if the way the scripts are made now, they generate the silkscreen marking the "highest" pin number instead of pin number 1.

Instead KLC states it should look somewhat like this:
[image]
according to http://kicad-pcb.org/libraries/klc/F5.1/

If need be, if have the fix ready for a PR. Although it is reasonably easy to fix (two minor changes).

evanshultz, 2019-01-28:
There are a few different versions that have been hand-made, while newer footprints used the scripted style. Eventually we should probably align the styles but as long as pin 1 is marked in some fashion (KLC may not show every single style) it should be OK.

JacobEFO, 2019-01-28:
It's very ambiguous that you can end up with both types of footprints (silkscreens) on your PCB i'd say. It very much eliminates the reason for using silkscreen in the first place, since you will no longer be able to identify, which sort of silkscreen is used.

evanshultz, 2019-01-28:
While nobody is disagreeing that all footprints should share a common style, many older (hand-built) footprints have not yet been replaced by scripted versions (where possible) at this time.

evanshultz, 2019-01-28:
I'm not sure exactly which part of the image you're referring to, but I guess it's this:
[image]

Sometimes you will get this kind of result. See https://github.com/KiCad/kicad-footprints/pull/995.

I think I see what's going on. I didn't check the scripts myself just now, to be clear, but I think pad length is the key; if the pads are short compared with the body edge pin 1 gets the silk line shortened while if the pads stick out (so the PCB fabricator and eyeballs can clearly show they're different) the pin 1 silk may be longer. The script is not marking the highest pad number as removing silk is just as valid as adding it for a pin 1 mark.

What I'm getting at is that touching the script may not be the right call. KLC is a reference or guideline but it does not mean any other variations are unacceptable. Or perhaps a change is warranted. For example, perhaps there are unwanted differences between the scripts (such as gullwing vs *FN).

JacobEFO, 2019-01-28:
What I mean is, that the script for DFN_QFN currently generates footprints that look like this:
[image]

Where the extended silkscreen covers pin 16 (right side) instead of pin 1 (left side).

evanshultz, 2019-01-28:
No. In the example above silk is being removed above pin 1 to indicate pin 1.

Removing silk to mark pin 1 is valid. For an example outside KiCad, from an industry standard that influenced our scripts, see http://www.ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf#page=16 which covers some stuff proposed for IPC 7351 revision C. The leftmost footprint on page 19 shows an extended silk line by pin 1, but that's only possible because the pads stick out well beyond the body edge.

(I assume you must mean pin 6 and not 16.)

JacobEFO, 2019-01-28:
Yes, I did in fact mean pin 6, I have recently been buggering around with 16-pin packages.

Oh I absolutely understand your point now. Since DFN is a bottom lead package the missing silkscreen marks pin 1. 

I've been using the same presentation as guidelines for my own footprints, I never realized the specific difference between bottom leaded components.

JacobEFO, 2019-01-29:
I actually do have more questions regarding the silkscreen. 

I see many more DFN footprints marking pin 1 with the extended silkscreen than using the missing silkscreen. Since KLC does not explicitly state which convention should be used (apart from: _Pin-1 designator is provided on the F.SilkS layer_), I think KLC should clearly state the convention on this.

Right now people will find DFN components that use either convention, while most of the gullwing components follow the traditional pin 1 extended silkscreen convention.

poeschlr, 2019-05-29:
Sorry for the late answer. Both the new dfn and gullwing definitions follow current industry standards which state that no lead packages use missing silk and gullwing uses extended lines.

Sadly we did not yet get around to convert all dfn/qnf parts to be script generated. (More work than you would think.) This means this is simply ongoing work.

fenugrec, 2020-04-25:
I just got bit by this on a DFN-6 package. Judging by the number of open and closed issues both here and on the kicad-footprints repo, I think this is a real problem, but I do understand the challenge in reconciling scripted- and hand-made footprints.
I suggest in the meantime the KLC could be clearer - the only hint I saw in KLC 5.1 is `Pin-1 designator must be visible after board assembly`. Not very specific; I had to come here to read

>  current industry standards which state that no lead packages use missing silk 

I wasn't able to find a document showing this, IPC7351 from 2005 doesn't seem to mention it, and the link to `http://www.ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf#page=16` posted earlier here is now dead.

I'd argue that the whole "missing-line silk" idea is completely dumb but I am currently a bit biased. I could settle for "counter-intuitive" or "inconsistent"  : )

poeschlr, 2020-04-25:
For gullwing parts it is easy. The pads are outside the package and therefore there is enough space to place a long additional line that is easy to see. If you check the dfn footprints that tried to emulate the gullwing option then you will find that the longer line is barely longer than the short one so it is really near impossible to see. Especially as the standards now reduce the pad size outside the package even further.
Even worse are LGA packages whose pads do not prodrude at all so there is even less of an option to have the longer line.

This is exaclty why IPC decided to go with the missing line and is also the reason why we will use that as well.

Plus the screenshot in the KLC very clearly shows a DFN and a QFN with missing line i do not know how much clearer we could make it than show an example.

fenugrec, 2020-04-25:
Thanks for the reply. I understand the longer line method being impractical for the smaller DFNs. But why not a corner mark, dot, triangle, or any other unmistakeable symbol ? i.e. *anything* more obvious than changing line lengths. If a single dot is ok for some 0402 diodes, surely we can find space for something more conspicuous on DFNs ?

A random half-baked idea:
[image]

The LGAs you mention, and QFNs, are not as problematic IMO, because there's a graphic element in each corner, and it's immediately obvious which one is unique. Even that would be an improvement over the longer/shorter line.

As for the image in KLC, yes I saw it, but there's no corresponding written item, and isn't helped by the fact that the current library has both missing- and long-line styles for many DFN sizes. So if I'm adding another package because of a weird thermal pad, and I base it on the "wrong style", it adds to the chaos.

So, my points are : 
- if the KLC absolutely wants to rigidly follow IPC, and IPC insists on this silly missing-line business, then it would help to clarify this specific case, especially while the library is not uniform; and bonus for referring to the specific document (is it available online ?).
- if the KLC is OK with dropping the missing-line style, there are so many ways to make the pin1 mark more obvious.

poeschlr, 2020-04-25:
Your diagonal is counter productive as silk under the part negatively impacts yield. And the L around the corner does not work for high density designs. Really there is no need for us inexperienced users to come up with a standard when there is a perfectly good industry standard out there. Which is why we decided to go with the IPC suggestion.

fenugrec, 2020-04-26:
Wow, thanks for taking the time to unify those footprints. Didn't expect to see that so soon.

I know it's too late now, but it's interesting how altium's "IPC-compliant generator" goes for a single dot:
https://www.altium.com/documentation/sites/default/files/wiki_attachments/294211/IPC%20Wizard%20-%20DFN1_AD17.png

chschlue, 2020-04-26:
I find it interesting how far from the body that dot is.
A couple of those packages in different orientations and you just have useless dots all over the place.

evanshultz, 2020-04-26:
There are been different versions of IPC 7351. Elements have changed over time to suit new packages coming out and the way boards are being designed. I can see in the screenshot that the pad shape is rectangle which makes me think the Altium generator adheres to, or in that screenshot is configured to make, IPC 7351 (with no suffix) footprints. That standard was released in 2005. Things have changed dramatically in the last 15 years with respect to SMD packages so this isn't a valid argument in favor of dots.

Currently, only the simplest boards my company designs could use silk outside the part body for marking pin 1. And this goes for more than just small no-lead packages. There are many package styles where removing silk is the most clear way to show pin 1 on an actual board. BGA and LGA are other package styles.

It can be an adjustment to see a new way of marking things, and I get that different package types may be more or less affected by this, but removing silk isn't forward-thinking. It is the best way to mark pin 1 in the high-density designs being done today. And for several years in the past. It is sad that https://ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf isn't working to give some good examples; I thought I would have downloaded it but I can't find it now. And note that this was a target for IPC 7351 C from 2015 so even it is 5 years old!

@fenugrec 
I don't mean to be dismissive. But I believe we are following the industry with these footprints. If there are parts of KLC which would make this more clear would you consider submitting a PR to https://github.com/KiCad/kicad-website? What is intuitive is subjective so if you take the time to document things in a way that's most clear to you it will be easiest to incorporate into KLC and hopefully will help others too.

closed 2020-04-26 by evanshultz

fenugrec, 2020-04-26:
I rest my case. Thanks for the detailed reply @evanshultz ; in the meantime I was able to find that "what's new" document from the webarchive, http://web.archive.org/web/20190712122301if_/http://ocipcdc.org:80/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf .

> but removing silk isn't forward-thinking. It is the best way to mark pin 1

heh, I think there's an extra negation in there : )