pointhi/kicad-footprint-generator #303: Package_SO script generates bad 3D paths

antoniovazquezblanco, 2019-02-27:
Existing SO package generator includes mask information in 3D path. See https://github.com/KiCad/kicad-footprints/pull/1441

antoniovazquezblanco, 2019-02-27:
Also, footprints in the official repo seem to be outdated compared with this script. After fix, all packages should be generated again.

evanshultz, 2019-02-27:
https://github.com/KiCad/kicad-footprints/pull/1371 will update a bunch of non-EP SO footprints with scripted ones. But that won't affect any ones with special mask requirements.

poeschlr, 2019-05-29:
The mask info is in there on purpose as the copper pad is not really what determines the size that will be found on the package itself. (The mask property is only shown if the EP is soldermask defined. Our naming scheme says that the EP parameter for the footprint is the copper size. The convention of giving the footprint and 3d model the same name might not be fitting for EP parts. I will open an issue about this over on the footprint repo for discussion for v6.)