pointhi/kicad-footprint-generator #308: PowerPAK 1212-8 Single Don't connect to Zone

jneiva08, 2019-03-07:
I'm trying to connect the Vishay_PowerPAK_1212-8_Single Drain Pad (custom pad) to one zone, but it never connects.

After to spend some time to understand why it didn't work, decided to edit the footprint and see the definition on "Local Clearance and Settings".
On Connection to Cooper Zones, the Pad connections is None. It only have two options None or Solid.

If go to normal pad without having Custom Shape says "From parent footprint".

[image]

How can fix it?

evanshultz, 2019-03-07:
@jneiva08 
Unfortunately, the custom pad to zone connection support in Pcbnew leaves something to be desired. Thermal spokes are not currently possible, leaving only no connection (which obviously doesn't work) and solid copper connection.

You will need to change the Pad Connection to Solid in order for a custom pad to connect to a zone. I have done this for many footprints at https://github.com/KiCad/kicad-footprints/pull/981, but a few aren't included (including this one). I just submitted https://github.com/KiCad/kicad-footprints/pull/1470 to fix the zone connection for this footprint and the dual version.

There is a bug report about this at https://bugs.launchpad.net/kicad/+bug/1732720.

jneiva08, 2019-03-10:
Thanks @evanshultz for the information, hope it will be fixed soon.
Sorry for open in wrong place :(

Will close this issue in the next days

closed 2019-03-12 by jneiva08