pointhi/kicad-footprint-generator #317: ipc_noLead_generator.py : Non-ASCII character '\xe2'

aewallin, 2019-03-19:
I get:
$ python ipc_noLead_generator.py 
  File "ipc_noLead_generator.py", line 50
SyntaxError: Non-ASCII character '\xe2' in file ipc_noLead_generator.py on line 50, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

seems to be associated with square-root symbols in the source file.
would be nice if this script would work on a standard vanilla ubuntu/python install...

evanshultz, 2019-03-19:
Most of the scripts are now written targeting python 3, as can be seen at the top of that script. This is a known issue. I just ran it this morning on 3.7.2 and it worked for me so I believe the script is fine.

aewallin, 2019-03-19:
then Readme should be changed to:
Supports: <strikethrough>Python 2.7</strikethrough> and 3.3+

aewallin, 2019-04-13:
the solution is to use python3, not 2. closing.

closed 2019-04-13 by aewallin