pointhi/kicad-footprint-generator #319: Trying to generate for TSM-SH series

gcormier, 2019-03-21:
https://www.samtec.com/products/tsm

@evanshultz Bringing it over here from the kicad-footprints issues.

I've slightly modified the existing Harwin script that you found for similar pins.

Note I'm using windows with the bash shell provided by Git.

Here's where I'm at just trying to get it to spit something out
```
Greg@MEDUSA /d/kicad/kicad-footprint-generator/scripts/Connector/Connector_Samtec (samtec-tsm) $ python conn_samtec_tsm_sh.py
conn_samtec_tsm_sh.py:200: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  configuration = yaml.load(config_stream)
conn_samtec_tsm_sh.py:206: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  configuration.update(yaml.load(config_stream))
Traceback (most recent call last):
  File "conn_samtec_tsm_sh.py", line 210, in <module>
    gen_family(configuration)
  File "conn_samtec_tsm_sh.py", line 190, in gen_family
    gen_footprint(x, mpn.format(pincount=x), configuration)
  File "conn_samtec_tsm_sh.py", line 80, in gen_footprint
    orientation=orientation_str)
KeyError: 'mounting_pad'
```

evanshultz, 2019-03-21:
Did you install a YAML parser? use `pip` to snag one. I think I used `pyyaml`.

Also, which version of Python are you using?

gcormier, 2019-03-21:
I ran `manage.sh update_dev_packages` which installed the required dependencies I think? In anycase, pyyaml is already installed.

Python 3.7.2

evanshultz, 2019-03-21:
Well dang. I get the same result. @antoniovazquezblanco ? I'm out of time today but can look at this tomorrow if Antonio doesn't answer.

Can you run other scripts? The other Harwin one doesn't work for me but `conn_jst_eh_tht_side.py` in the Connector_JST folder does.

I'm also on 3.7.2.

gcormier, 2019-03-21:
Yup, that one worked, although warnings about safe load, and also

`Invalid -W option ignored: invalid module name: 'yaml'`

gcormier, 2019-03-21:
#320 for when things are semi-functional.

Dunno how I feel about the detailed silk screen... also, Samtec offers 4 pin lengths, also unsure how to handle. Perhaps a graphical display with 4 lines indicating the different offsets? But keep the traditional/most common depth?

Quick example:

[image]

evanshultz, 2019-03-22:
Fixed both Harwin script issues at https://github.com/pointhi/kicad-footprint-generator/pull/322.

On pin length, I have two proposals. Our library is big enough to support many footprints, but since it's also general-purpose I'm inclined to support the greatest number of footprints possible.
1. Just make the one you want.
1. Build the longest one and then shorter ones will fit inside. This is the lowest common denominator approach. I favor this for our library if it works for you.

gcormier, 2019-03-22:
Sweet, that'll get me tinkering.

My issue with option 2 is that I bet 99% will use the standard length. Building the longer one will waste space for most people designs.

Another thought:

This connector would very likely be at the edge of the board, or if not, it would need to have quite a bit of space left in front to allow for plugging in of the header.

The pin outline could reflect the standard pin length, but the courtyard could reflect the longest pin length? For something overhanging the edge of the board, the courtyard would have no ill effect. Of course it would violate KLC, which might bring us back to option 1 which is the one I want :)

gcormier, 2019-03-22:
Okay, so I had an issue with the pin 1 silkscreen.

[image]

I changed a few lines, from 6.775 to 6.98

```
	s_pin1 = [
        {'x': -6.98+padsize[0]/2-(2.5/2+configuration['silk_pad_clearance']+configuration['silk_line_width']), 'y': -(pitch*(pinnum-1))/2},
        {'x': -6.98+padsize[0]/2-(2.5/2+configuration['silk_pad_clearance']+configuration['silk_line_width']), 'y': -(pitch*(pinnum-1))/2-1/2-configuration['silk_line_width']-configuration['silk_pad_clearance']},
        {'x': -6.98+padsize[0]/2, 'y': -(pitch*(pinnum-1))/2-1/2-configuration['silk_line_width']-configuration['silk_pad_clearance']},
	]
```

Looking better. That number 6.775 is everywhere. And then I noticed the silkscreen for the pins is off as well.

[image]

So it leads me to believe perhaps I need to be fixing it elsewhere? Or is this just part of the game?

I do notice a silkscreen pad offset, but can't find that configured anywhere, and perhaps I don't want to touch that.

gcormier, 2019-03-22:
For reference
[image]

gcormier, 2019-03-22:
A few tweaks, latest commit has this coming out the other end

[image]

gcormier, 2019-03-25:
Thoughts on this? Should I submit a PR for continued discussion in kicad-footprints ?

evanshultz, 2019-03-27:
The courtyard should match the footprint.

Good catches on the silkscreen. Magic numbers are no good, especially if it's used all over the code. You will want to keep 0.12mm or more clearance from pads to silk, which turns into 0.18mm since you need to consider half of the silk line width (or the endcap radius). If there is a silkscreen pad offset not being used, it seems like a good variable name so put it to use or remove it so nobody else is confused in the future.

Script quality can vary, and the framework sometimes changes, so it may take a bit of work to start with an existing script and end up with something nice. Great work!

The screenshots above look OK, but in this case a footprint PR with all the parts to view in KiCad and a script PR will be the best way to get this reviewed and merged.

gcormier, 2019-03-27:
I'll get a PR going.

RE: Courtyard, I followed KLC for a connector. One might argue that the pin-side should be even bigger to allow mating?

> Connectors should have a clearance of 0.5mm, in addition to the clearance required for mating of connector