pointhi/kicad-footprint-generator #341: repeated thermal_vias statement?

aewallin, 2019-04-12:
http://www.ti.com/lit/an/slma004b/slma004b.pdf
For this footprint one would want to repeat the thermal_vias statement.
one set of vias in the un-masked exposed-pad area (small diameter and dense grid)
another set of vias in the masked pad-area (large diameter and large grid)

is this possible currently? (I could not get it to work).
thanks.

poeschlr, 2019-05-05:
Yes footprints like that can be generated using the ipc generator scripts for gullwing parts.

See for example HTSSOP-16-1EP_4.4x5mm_P0.65mm_EP3.4x5mm_Mask2.46x2.31mm_ThermalVias which is script generated.

The class that handles this is the exposed pad node found in the specialized node section.

aewallin, 2019-05-06:
The HTSSOP-16-1EP has a large EP-size with a smaller opening in the mask - this is OK. I think I am doing it similarly in this PR: https://github.com/KiCad/kicad-footprints/pull/1547

My question was on the suggested TI-footprint which has two different via-grids. One central (3x5 holes, 13-mil diameter) with small holes, and another with 2x3 larger (25mil diameter) vias.

Do you know if the two different via-sizes + grids can be scripted?

[image]

evanshultz, 2019-05-06:
No, this is not currently possible.

Since the yellow is optional and the gray is required, how about scripting the gray part? If the user wants more for their specific application they can make this a board-level (not footprint-level) change.

aewallin, 2019-05-06:
yes, they gray parts are PRs:
https://github.com/pointhi/kicad-footprint-generator/pull/340
and
https://github.com/KiCad/kicad-footprints/pull/1547

poeschlr, 2019-05-10:
It would definitely be possible to develop a script specifically for this footprint but i would not suggest to add such specialized things to the generic generators. The question then is how useful such an extended script is as it is for a very low part count.