pointhi/kicad-footprint-generator #375: Hirose DF63 1Pin connector missing NPTH

jhalmen, 2019-06-11:
The `Hirose_DF63-1P-3.96DSA_1x01_P3.96mm_Horizontal` footprint does not include the NPT hole as the connectors with more pins do. This is erroneous. It should include it, as the [datasheet](https://www.hirose.com/product/document?clcode=CL0680-0529-2-00&productname=DF63-1618PCF&series=DF63&documenttype=Catalog&lang=en&documentid=D31622_en) (page 3), but also the following picture, show.
[image]

Additionally, all the `DF63` connectors currently in the library are named `Horizontal`, although they're actually the vertical parts. (their 'angled' counterparts would be the horizontal ones).

Would a patch be accepted, that adds the NPTH unconditionally (removing this [line](https://github.com/pointhi/kicad-footprint-generator/blob/4b4fcc8c5a8ee948747c915815fae72e64c40e91/scripts/Connector/Connector_Hirose/conn_hirose_df63_tht_top.py#L150))?

evanshultz, 2019-06-11:
Yes, that does look correct to have the peg on the single pin connector. Removing the line you indicated (and moving the following line back an indent level) should correct this. Good catch and thanks for taking the time to report it!

Regarding the orientation, I believe this was flipped in an external file and corrected some time ago. Just generate the new files and it should hopefully be fixed already.

However, something more insidious lurks: The peg position is determined by the connector being DF63M or DF63R. This distinction is lost on the current script. It's unclear to me from the datasheet if the script builds the M or R version, but that needs to be corrected as well. We will want to leave the existing connectors for now, along with the new ones, and we can clear the old ones out later.

Lastly, another error is that line 97 specifies the wrong manufacturer. Could you correct that as well?

jhalmen, 2019-06-11:
The script builds the M version. Source: i bought the M-version and it fits (except for the 1Pin connector).
Not having looked too much at the way the generator works: how do i best make the distinction between the R and M version?
Or will you handle that part personally?

PR #376 (2019-07-09): Hirose df63
fixes #375
closed 2019-07-09 by pointhi