pointhi/kicad-footprint-generator #395: Inconsistent return values from getRealPosition methods

calebreister, 2019-07-18:
I've been working on a fairly complicated [footprint generator script](https://github.com/calebreister/kicad-footprint-generator/blob/CalebReister/scripts/Connector/Connector_Samtec/QStrip_Vertical.py). Part of this script utilizes the tree shown below.
~~~
+ Node
  * Translation [x: -27.25, y: -2.485]
    * Rotation [r: 180]
      * PolygoneLine [[x: -0.5, y: 0.5] ,[x: 0.0, y: 0.0] ,[x: 0.5, y: 0.5]]
        * Line (fp_line (start -0.5 0.5) (end 0 0) (layer F.Fab) (width 0.1))
        * Line (fp_line (start 0 0) (end 0.5 0.5) (layer F.Fab) (width 0.1))
~~~
However, attempting to rotate the `PolygoneLine` object produced an `AttributeError`:
~~~
Traceback (most recent call last):
  File "./QStrip_Vertical.py", line 300, in <module>
    generate_one_footprint(fp_params, config, args.library)
  File "./QStrip_Vertical.py", line 246, in generate_one_footprint
    KicadFileHandler(fp).writeFile(filename)
  File "kicad-footprint-generator/scripts/Connector/Connector_Samtec/../../../KicadModTree/FileHandler.py", line 54, in writeFile
    output = self.serialize(**kwargs)
  File "kicad-footprint-generator/scripts/Connector/Connector_Samtec/../../../KicadModTree/KicadFileHandler.py", line 103, in serialize
    sexpr.extend(self._serializeTree())
  File "kicad-footprint-generator/scripts/Connector/Connector_Samtec/../../../KicadModTree/KicadFileHandler.py", line 143, in _serializeTree
    sexpr.append(self._callSerialize(node))
  File "/home/caleb/GitHub/kicad-footprint-generator/scripts/Connector/Connector_Samtec/../../../KicadModTree/KicadFileHandler.py", line 161, in _callSerialize
    return getattr(self, method_name)(node)
  File "kicad-footprint-generator/scripts/Connector/Connector_Samtec/../../../KicadModTree/KicadFileHandler.py", line 220, in _serialize_Line
    sexpr += self._serialize_LinePoints(node)
  File "kicad-footprint-generator/scripts/Connector/Connector_Samtec/../../../KicadModTree/KicadFileHandler.py", line 211, in _serialize_LinePoints
    ['start', start_pos.x, start_pos.y],
AttributeError: 'tuple' object has no attribute 'x'
~~~

After looking at the inner workings of KicadFileHandler.py, Translation.py, and Rotation.py, I found the source of the error: `Rotation.getRealPosition()` *always* returns a tuple, and many of the serialization methods expect `Node.getRealPosition()` to return a `Vector2D` object. However, if a rotation argument is included, `getRealPosition()` functions generally return a tuple containing the position and an angle in degrees.

For now, I made a [minor modification](https://github.com/calebreister/kicad-footprint-generator/commit/e7fe7ecfb343414a9450a262d37f502e9741d894) to `KicadFileHandler._serialize_LinePoints()` that resolves my issue. However, it would probably be a good idea to make all `getRealPosition()` methods return a single type. I see several ways to do this:

- Change all `rotation=None` default parameters to `rotation=0` and modify all `_serialize` methods to accept a tuple from `getRealPosition()`.
- Several classes redefine `getRealPosition()` to return a `dict`. For instance, `Translation` includes this definition:
   ~~~
    translation_coordinate = {'x': parsed_coordinate.x + self.offset_x,
                              'y': parsed_coordinate.y + self.offset_y}
   ~~~
   Adding an 'r' or 'theta' element to such a dict (even if it is zero) would be inconsequential in cases where only the coordinates are required (`Vector2D` ignores the extra element), and it should be fairly easy to modify the `_serialize` functions that require the rotation.

pointhi, 2019-07-18:
I think the cleanest solution would be using a transformation matrice. This should remove quite some of those hacks.

calebreister, 2019-07-18:
@pointhi How would that affect backward-compatibility? I don't know much about the inner workings of the API, but there are definitely a variety of things that could be cleaned up.\* However, it is *much* easier to use than the KiCad Python API. If a rework is in order, it may also be a good time to add more kwargs and provide better documentation for the existing ones. Here are a few ideas/suggestions:
- Merge the filled and line shape functions (example: merge `PolygoneLine` into `Polygon` with a `fill=True` kwarg).
- Provide an option to make a mirrored copy of a shape.
- Create a `Transform` node capable of translating, scaling, and rotating child nodes

If you plan on performing matrix operations, have you considered adding numpy as a dependency?

\* For instance, what is a "polygone?"

calebreister, 2019-07-24:
@pointhi I can't make any guarantees, but I'm willing to try reworking some of the API code. It would help to have some input on my ideas and an answer to the numpy question.

poeschlr, 2019-07-31:
I have added translation and rotation functions for the vector classes in https://github.com/pointhi/kicad-footprint-generator/pull/351 (using transformation matrixes)

So you might not really need to do much but call these functions on all points of the node. (which i also did for a lot of primitives already in that pull request.)