pointhi/kicad-footprint-generator #420: Chip body sizes are too generic?

evanshultz, 2019-09-06:
@poeschlr 
A few things with https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/SMD_chip_package_rlc-etc/size_definitions/size_default_chip_devices.yaml.

1. The tortai-tech.com link seems broken. I assume that was from IPC-SM-782? Perhaps a comment to indicate the document source, even if we can't find it available online?
2. SMD_0603 matches IPC-SM-782 sizes so I assume that was the reference. But for SMD_0805, the only significant deviation from the IPC doc is in terminator spacing. I suppose we could use the IPC spec for the other two dimensions and then just modify the terminator spacing to suit our needs (we both will recall the huge variation we found and the limits we placed on this parameter based on gathering data from parts in the market).
2. Here is the major issue that I found which caused me to make this issue: I'm concerned the generic SMD_0603, SMD_0805, etc. size definitions aren't specific enough. IPC-SM-782 shows different dimensions for chip resistors (page 72) and chip capacitors (page 76) which are NOT the same for the same body size. So instead of SMD_0603 should we have C_0603 and R_0603, and the same for other body sizes and part types?

If I'm right about #3, it means that our footprints are using correct IPC calculations for fillet goals, but the body sizes that are an input to the IPC calculations could (should?) be more specific for different part types.

poeschlr, 2019-09-07:
The 0805 was decided here: https://github.com/KiCad/kicad-footprints/issues/711

We can of course have them differ between different devices if they indeed are different.

evanshultz, 2019-09-07:
I didn't take the time to try and find the conversation since I figured we could both recall, but that is it. Thanks.

I suppose when reviewing IPC-SM-782 having unique dimensions for resistors and capacitors (and other package types) was missed. This is easy enough to update in the script. Do you recall if IPC-SM-782 was the only body size source (I know the Google doc massaged things a bit).

So
1. I'll search for another link, but definitely add a comment in the YAML file to tell users what the document is supposed to be.
1. If IPC-SM-782, or another standard doc, is close to the sizes found on the market I'll use it.
1. Since it appears having Rs and Cs the same was accidental when IPC gives unique dimensions, I can split the 'default' body size YAML file (linked above) into cap and resistor files and then call only the appropriate one from the YAML file in the folder above. Easy.

evanshultz, 2019-10-02:
I'm working on this now.

evanshultz, 2019-10-02:
PR at https://github.com/pointhi/kicad-footprint-generator/pull/438.