pointhi/kicad-footprint-generator #434: manage.sh gives "invalid syntax".

MathieuMH, 2019-09-25:
Hi all,

I'm new to the whole kicad footprint scripting. A couple of years ago I followed a course to learn the basics of python. Therefor I wanted to start creating my connector using the scripting tool.
On the launch page the is a way to install and update all depencies. When I excute this line I imidiatly get a invalid syntax message. Does anyone have an idea why?
Using python 3.7 on Windows 7 machine.
I ran a command windows in the kicad-footprint-generator-master, and typed:
`python manage.sh update_dev_packages`

The result is: 
`File "manage.sh", line 3
if [[ "$OSTYPE" == "darwin"* ]]; then

SyntaxError: invalid syntax`

evanshultz, 2019-09-25:
I would use `pip` to install any dependencies you need.

That being said, let's cut to the chase: What script are you trying to run?

tolot27, 2019-09-25:
`manage.sh` is a shell script, not a python script. Hence, you get the error message. You can run it only inside a bash shell like Git Bash.

MathieuMH, 2019-09-26:
@evanshultz
My ultimate goal is to make a script to generate all the Samtec Searay connectors.
So I was reading through the readme of this footprint generator and I just cloned the repository and was going to try an example to see the workflow. And so I was stuck.
At first I wandered why is was .sh instead of .py, but didn't give it a lot of thought.

The documentation is also not really clear on what to do. So I want to make a script for a connector series. What do I need?
- Python 3.5+ (I have 3.7.4 installed)
- Any other library I should install using PIP?

evanshultz, 2019-09-26:
First, we can help you get the scripts running, but we expect you to be able to develop the script yourself. The info here lightly describes the framework; it's not a tutorial. So you're largely going to be on your own. There is a lot of help online for writing Python so you'll be OK.

OK. So then let's start by running an existing script. Let's use https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Samtec/conn_samtec_hle.py. (Some of the other Samtec scripts have apparently become rotten or the framework has been changed in ways that cause them not to run, at least for me.)

First, we will need a YAML parser. I use `PyYaml` but there are other choices.

Open a shell (Command Prompt or PowerShell window in Win 7) and type `pip install pyyaml`. That should download and install a YAML parser.

Once that's done, browse to the folder of the footprint generation script above and type `python conn_samtec_hle.py`. A folder should appear in the directory with a pile of footprints in it.

Then you're ready to copy that python file and start modifying it to generate your new footprints or else start a new python file for your scripts but keep using the `KiCadModTree` framework. Either way works and there are a lot of various scripts to look at for inspiration.

Lastly, a `*.sh` file isn't a 'normal' Windows script. In Win7 you won't be able to run it. It is useful in a Linux OS. Actually, the error message you posted above explains. That was your initial issue.

MathieuMH, 2019-10-01:
evanshultz
Thanks for the info. I made a lot of progress with the script. I need to clear out some more details on the framework. But I'll manage.
The Samtec Searay are quite complex connectors. So I have a few more challenges to overcome.

matthijskooijman, 2020-04-29:
Isn't the original issue here resolved? Also, I noticed #515 that suggests converting manage.sh to python to make it easier to run, so maybe this issue should be closed?

Franck78, 2020-09-01:
I really see zero issue here.
You can close this, yes.
I hope you know now that a language source file is interpreted/compiled only by the interpreter/compiler it is written for !

closed 2020-09-02 by pointhi