pointhi/kicad-footprint-generator #465: Molex KK 254 series drill size too big (conn_molex_kk_254_tht_top.py)

HendryKaak, 2019-11-28:
Hello,

I've been using multiple pin variants of the molex KK 254 serie connectors on one of my PCB's, but when I ordered the PCB and fitted the connectors in it was possible to rotate the connectors in place quite much. Looking into the [code here](https://github.com/pointhi/kicad-footprint-generator/blob/5c5ec7177fc7ca256819aa412e4f261a7fc1bc72/scripts/Connector/Connector_Molex/conn_molex_kk_254_tht_top.py#L41) I found out that the drill size is oversized compared to the [KLC 7.6](https://kicad-pcb.org/libraries/klc/F7.6/), which describes a hole size of the maximum pin diameter + 0.2mm in size. I would expect the size to be set to ~0.84mm AFAIK since using a bigger hole size adds up an excessive amount of solder necessary for creating a proper connection and the rotation problem as well.
Also I couldn't find any reference about this "minimum 1.1mm drill size" (as mentioned in the comment of the previous code link) in the manual (maybe something of old times?), so is there another reason for making the hole diameter this big?

chschlue, 2019-11-28:
A 0.64 mm square pin has a diagonal of sqrt(2*(0.64 mm)²) = 0.91 mm.
The script is might be overzealous in rounding up to 1.2 mm though.

HendryKaak, 2019-11-28:
> A 0.64 mm square pin has a diagonal of sqrt(2*(0.64 mm)²) = 0.91 mm.

Ah yes good call. It's a square pin indeed.

>The script might be overzealous in rounding up to 1.2 mm though.

Yes exactly, looking at the following Molex KK254 connector [drawing](https://www.molex.com/pdm_docs/sd/022032021_sd.pdf) in the bottom left corner, they also recommend 1.02±0.05 which should be enough clearance for a pin of this size.

chschlue, 2019-11-28:
[Here](https://www.molex.com/pdm_docs/sd/022272021_sd.pdf) they recommend (1.19±0.0.5) mm.

HendryKaak, 2019-11-28:
>Here they recommend (1.19±0.05) mm.

So far the drawing consistency, [here](www.molex.com/pdm_docs/sd/1718560002_sd.pdf) is another one with a different recommendation being 1.14±0.05  :")

I'm thinking about measuring the pin itself tomorrow to see what the pin-size really is, since some of the drawings mention that the 0.64mm is already the max diameter of the pin (as seen in D11 of this [drawing](www.molex.com/pdm_docs/sd/022057028_sd.pdf); note the ø0.64mm).

HendryKaak, 2019-11-29:
Ok, I took some measurements today, sadly I don't have a caliper with a more precise 0.01mm scaling, but it is doable to see what the size is.
[image]
The first image is a little bit tilted, but shows that the side of the pin is indeed ~0.64mm (not in diameter ofc).

[image]
This image shows that the pins are not exactly square, but more like squared with rounded edges, since it measures as 0.85mm instead of the 0.64*sqrt(2)=0.91 mm. I have measured this with multiple pins and connector variants (2-pin and 8-pin) to verify that ofcourse.

[image]
[image]
These are pictures of the actual pcb with the 1.2mm holes, which shows that especially with 2-pins connectors that the amount of play is too much for keeping the connector nicely within the silkscreen courtyard.

Following the KLC by adding the 0.2mm margin, I would suggest to make the hole size 0.85+0.2=1.05mm.

poeschlr, 2020-04-26:
TlDr: We need to indeed fix the drill size. But we need to fix it to the datasheet suggested value of 1.19mm! (Rounding here is definitely not ok as it will increase the error if somebody orders at a board house that uses imperial dirlls)

---

Both datasheet you linked are for a different part number so they can be discarded (the other part numbers are expected to have different tolerance ranges).

---

You measured one device that can be anywhere in the tolerance range. You would need to measure a huge set of devices from different production runs to get any idea of the tolerance range. Plus your own measurement tolerances would also need to be taken into account (and i guess would need to be improved as the tolerances we want to measure are likely to be smaller than your measurement tolerances). So lets stay with the datasheet values.

---

The datasheet tells us the pins are 0.64 square (but does not state where in the tolerance range it is) which results in a diagonal of 0.91mm leading to at least a drill size of 1.11mm when assuming that 0.64 is indeed the maximum size for such pins and that there is no tolerance in pin placement. The 1.19mm size is therefore not really far off from what i would expect if there are even moderate tolerances here. Especially if the pin size is given as nominal (which i expect as normally in a drawing everything is nominal unless otherwise specified).

---

Remember the KLC definition comes from IPC which is not intended for a tight fit between the part and its leads. There is meant to be ample space for solder to properly fill up the hole without voids. If your process requires tighter holes for alignment reasons then this guideline will not apply to you. This is ok. Technology is full of tradeoffs!

---
---

In your case i suspect that you had bad luck. The parts you got will likely be on the low end of the pin size tolerance range. And the drill might be on the large end (compounded by the fact that we already started with a 0.01mm increased hole size assuming your manufacturer even has a 1.2mm drill in stock. If you ordered at an imperial board house than the target hole size might be even larger than the 1.2mm requested by the footprint)

chschlue, 2020-04-26:
> In your case i suspect that you had bad luck. The parts you got will likely be on the low end of the pin size tolerance range. And the drill might be on the large end (compounded by the fact that we [...]

Don't forget that these are plated holes, so there's more to it than just whether they happen to have a 1.2mm or 3/64" drill.
But I like the term "imperial board house". Sounds like Darth Vader's Motel.

poeschlr, 2020-04-26:
With drill i mean "target final hole size". I was just lazy when writing the above.

closed 2020-04-26 by evanshultz

HendryKaak, 2020-04-28:
> We need to indeed fix the drill size. But we need to fix it to the datasheet suggested value of 1.19mm! (Rounding here is definitely not ok as it will increase the error if somebody orders at a [...]

@poeschlr  Now that I'm looking at this, I see where the problem is why I'm having so much difference in drill size compared with the datasheet @chschlue showed before. I'll come to that in a moment.

> You measured one device that can be anywhere in the tolerance range. You would need to measure a huge set of devices from different production runs to get any idea of the tolerance range. Plus your [...]

That is not entirely true, I did measure multiple of the connector series (and also from 2 different orders), but it's hard to tell if it's still the same batch. If measurements are off way more than the 5% of the datasheet, there must be something else wrong. I'm ok with staying close to the datsheet measures, because of course only the manufacturer can know of the tolerances and sizes (also because of the imperial measures as stated before).

> The datasheet tells us the pins are 0.64 square (but does not state where in the tolerance range it is) which results in a diagonal of 0.91mm leading to at least a drill size of 1.11mm when assuming [...]

Well I can assure you that 0.91mm with hole sizes of 1.19mm is really overzealous, that's actually why I posted this issue with close-up pictures in the first place. I even ordered a second batch of PCB's with drill size ~1mm (same as the default 2,54mm pin headers in Kicad) and they were perfect, not overzealous or too small at all.

> Remember the KLC definition comes from IPC which is not intended for a tight fit between the part and its leads. There is meant to be ample space for solder to properly fill up the hole without [...]

I'm perfectly aware of that, I'm don't even prefer tight fits, but when using connectors, it should be rotating on the board in all kind of directions like they do now.

> In your case i suspect that you had bad luck. The parts you got will likely be on the low end of the pin size tolerance range. And the drill might be on the large end (compounded by the fact that we [...]

I thought so as well until a few batches (PCB's and connectors) still had the same problem.

----

Ok and here comes the part where I must apologise for the confusion. I have misinterpreted the product KK254 as the series naming of the product I've been using, which is not the same as the script implemented series (just 6410). The ones I've been using is the product KK254 6373 series, which have a drill size of 1.02mm specified in the datasheet [here](https://www.molex.com/pdm_docs/sd/022112022_sd.pdf).
I would like to add those connectors to the series if possible, but the `conn_molex_kk_254_tht_top.py` script might need some refactoring compared to other scripts. I'm not sure which I should use as reference. Could you (@chschlue ?) perhaps point me to one of the latest script versions for that please?

So again, sorry for the confusion that was definitely my fault and thanks for the help :+1:

evanshultz, 2020-04-29:
@HendryKaak 
The script that Rene updated is recent, so feel free to build of off https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Molex/conn_molex_kk_254_tht_top.py or copy it. What we have is a bit of a mix and sometimes a single script could work for an entire family but there are separate scripts anyway. Generally we will take whatever the contributor chooses since they were willing to put in the work.

HendryKaak, 2020-04-29:
OK, will do. Thanks

HendryKaak, 2020-05-11:
Hello again,
I've been in contact with Molex support regarding the drill size recommendations in the datasheets. Since I was curious about the difference in recommendations for the KK 254 6410 Series and the 6373 Series, which seem to have the same pin size. I'll leave the mail conversation down below for future reference, especially because it shows that datasheet recommendations might not always be ideal.

------------------
> Hello Molex support,
I have a question regarding the product KK254 6410. I've been making a PCB footprint from
this connector (for open source usage) and stumbled upon some peculiarities.
The drawing of this connector describes a drill size of about 1.19mm/.047". Which I've already
included in the footprint and made PCB's from. But after checking the final product, I saw that
the connectors fit in very loosely inside their THT holes (almost to the point where you can tilt
the connector horizontally).

> Measuring the pin diameter, it is about ~0.94mm in diameter which is a diameter size difference
of almost 0.25mm and that is well overzealous if you ask me. I also checked the mounting holes
in the PCB and they perfectly match the 1.19mm. We make use of multiple of the same series with
2 to 8 pins variants and I measured multiple pins to make sure it wasn’t just a batch mistake.
Connector KK254 6373 seems to be almost the same (except for the plastic friction lock size) and
that one has recommended drill size of 1.02mm/0.04".
Now the question is of course, what is the reason for this? Or is this a PCB recommendation error?

> Btw, replies in Dutch is OK as well. This is what I sent to the American Molex support before they
redirected me to this email address.
Met vriendelijke groet / With kind regards,
Hendry Kaak
----------
>Hello Hendry,
Thank you for bringing this up.
The recommendation we have on 6410 with drill-size of 1,19mm +0,05 is based on experience thousands of customers have and we also have tested many years ago.
Normally with the diameter of 0,94mm what you have measured it should fit.
We have a difference in the recommendation for 6373-family because both series from the same family have been designed many years ago in different regions. 

> Both should fit, and both are only recommendations (see “typ” behind the value). 6373 is a little bit tighter (6419 is maybe better suitable for automatic placement on the PCB), but with the normal soldering-process (wave) both should not be a problem. Did you measure your tolerance?
Thanks and BRGDS
-----------------
> Hello Molex support,
Thank you very much for replying to my email.

> “We have a difference in the recommendation for 6373-family because both series from the same family have been designed many years ago in different regions.”
OK, at least that explains why most of the drawing in the kk254 product range have different drill size “recommendations”.
Wouldn’t it be a good idea to rectify all those different recommendations to the current IPC2222 standard with the class type (1,2,3)? 
Or at least mention which standard is used for the current recommendations?

> “Both should fit, and both are only recommendations (see “typ” behind the value).”
I know that the recommendation fits (in some way or another) but compared to the other family it is quite oversized.
And for production a lot of extra solder tin is needed to fill up the gap between the THT pad and the Molex pin. The pins with the plastic housing of the connector is also able to rotate a lot (this can be seen by the outlined silkscreen around the connector that we use). Which doesn’t make it a good fit for the TPA (mostly for 2/3 pin variants).

> “Did you measure your tolerance?”
Which tolerance are you referring to exactly? The pin tolerance?
I looked at the drawings from the 6410 family, but I must say that the drawings can be quite challenging to understand sometimes.
I could not seem to find the pin diameter tolerance. What I would expect from the drawings is that the drill size diameter is recommended at pin diameter 0.64mm[square single side] * sqrt(2) = 0.905mm + 0,2mm (following IPC-2222 class 2 for example).
Which is a drill diameter recommendation of about 1.11mm. The final plated hole will of course be slightly smaller than that.

> The main reason for pointing this out is because these (seemingly small) drawing errors lead to at lot of confusion and wrong footprint designs for PCB’s. Most of the time people tend to take your recommendations for **mandatory** instead of recommendation, because “the manufacturer probably knows what is best”. PCB software tool design guidelines also state in most cases that if the calculated sizes deviates from the datasheet, the datasheet values **must** be used instead of the calculated values.

> Convincing people to do otherwise takes a lot of extra effort, which should not be needed in most cases.
And if drawing a recommendation is not following a standard, maybe it is even better to **not** show the recommendation at all, so that people need to calculate their own drill size (or just point to a commonly used standard instead).
Met vriendelijke groet / With kind regards,
Hendry Kaak
---------------
> Hello Hendry,

> I fully understand your frustration. But there are simply some limitations of changes in specs and drawings.

[... cut ...]
> There is no error in the drawing, this is the way thousands of customers worked with since 50 years. Sorry, I know this is not a good argument, but I am working with Molex since more than 20 years, and I never faced this problem before. 

> I am sorry if you have problems with these products, but I need to tell you we cannot change anything here. 
Have a nice weekend.
bRGDS
--------------------
>Ok, thank you for your honest answers.
I’ll leave it at that.

>Have a nice day.
Met vriendelijke groet / With kind regards,
Hendry Kaak

PR #538 (2020-04-26): Fix wrongly rounded hole size of KK-254
Fixes #465

Footprint pull request https://github.com/KiCad/kicad-footprints/pull/2246
commit 880c84ad (2020-04-26): Fix wrongly rounded hole size of KK-254 (#538)

Fixes #465