pointhi/kicad-footprint-generator #551: Define center pad paste pad grid

matyas1995, 2020-05-03:
Hello,

I'm trying to script a VQFN package as described [here](https://www.nxp.com/docs/en/package-information/98ASA00694D.pdf).

There NXP chip this footprint belongs to has no single center pad, but 16 individual pads instead, each connected to GND: https://www.nxp.com/packages/SOT1649-1

Therefore, the position of the paste pads on the single large pad is fairly important for the soldering result. I have managed to place the thermal vias as recommended but fail to align the solder pads with the package center pins.

Is there an option for this in the scripts? I'm using the scripts at scripts/Packages/Package_NoLead__DFN_QFN_LGA_SON

Thanks,
Matyas

chschlue, 2020-05-03:
This package is basically half-QFN, half-LGA but the LGA part is sort of an NSMD EP with many small mask cutouts.
Not sure if this can (or should) be scripted at this time.

matyas1995, 2020-05-04:
Okay thanks for the help :)

Is the DFN_QFN Library still the right one to place such a footprint or should I rather place it into the Package_LGA library?

chschlue, 2020-05-04:
Yes, I'd say it's definitely closer to QFN than LGA.
NXP calls it QFN, too.

closed 2020-05-04 by matyas1995