pointhi/kicad-footprint-generator #556: documentation for footprint generation, fix pad size?

rs-development, 2020-05-18:
I am trying to create a footprint with the generator scripts.

https://www.diodes.com/assets/Package-Files/W-DFN3030-10-Type-TH.pdf

I am using the ipc_noLead_generator.py but i am missing a definition for fixed pin sizes to match the given landing pattern. How would you generate such footprint?

```
DFN-10-1EP_3x3mm_P0.5mm_EP1.65x2.6mm:
  device_type: 'DFN'
  manufacturer: 'Diodes Inc.'
  part_number: 'DGD05473'
  size_source: 'https://www.diodes.com/assets/Datasheets/DGD05473.pdf'
  # ipc_class: 'qfn' # 'qfn_pull_back'
  #ipc_density: 'least' #overwrite global value for this device.
  body_size_x:
    nominal: 3.0
    tolerance: 0.1
  body_size_y:
    nominal: 3.0
    tolerance: 0.1
  overall_height:
    nominal: 0.75
    tolerance: 0.05

  lead_width:
    nominal: 0.3
    tolerance: 0.05
  lead_len:
    nominal: 0.4
    tolerance: 0.1

  EP_size_x:
    nominal: 1.55
    tolerance: 0.05
  EP_size_y:
    nominal: 2.5
    tolerance: 0.1

  EP_size_x_overwrite: 1.65
  EP_size_y_overwrite: 2.6

  EP_num_paste_pads: [2, 2]

  thermal_vias:
    count: [2, 3]
    drill: 0.2
    # min_annular_ring: 0.15
    paste_via_clearance: 0.1
    EP_num_paste_pads: [2, 2]
    #paste_between_vias: 1
    #paste_rings_outside: 1
    #EP_paste_coverage: 0.7
    #grid: [1, 1]
    # bottom_pad_size:
    paste_avoid_via: False

  pitch: 0.5
  num_pins_x: 0
  num_pins_y: 5
```

chschlue, 2020-05-18:
Don't.
The whole point of the generator is generating IPC land patterns from package dimensions, not replicating suggested land patterns.

rs-development, 2020-05-18:
Oke that's fine. So am am using it just for a template generation.

closed 2020-05-18 by rs-development

evanshultz, 2020-05-18:
See https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Packages/Package_Gullwing__QFP_SOIC_SO for the gullwing generator which should help since it's a very similar script. A doc for the no-lead generator is being reviewed now.

rs-development, 2020-05-28:
I am reopening this, becouse of poeschls comment on the footprints PR

https://github.com/KiCad/kicad-footprints/pull/2290#issuecomment-633941930