pointhi/kicad-footprint-generator #582: Error in yaml

jneiva08, 2020-08-11:
I like to know if someone can help-me

I'm trying to execute the following script `./conn_molex_micro-fit-3.0_tht_top_dual_row.py` and it is giving error

`Traceback (most recent call last):
  File "./conn_molex_micro-fit-3.0_tht_top_dual_row.py", line 26, in <module>
    import yaml
ImportError: No module named yaml`

when try `./conn_molex_micro-fit-3.0_tht_side_dual_row.py` it works

I'm using linux

evanshultz, 2020-08-12:
They both run for me and generate footprints. And they both have `import yaml` on line 26. In fact, the header section is identical in both files. This seems a bit strange to me.

I'm on Windows 10 with Python 3.7.2 and the PyYAML 3.13 module installed.

Do you see this issue with any other scripts? Not sure this helps, but let me know if there's some info I can provide.

pointhi, 2020-08-12:
@jneiva08 you need to install `pyyaml`, either with the packet manager or using pip.

jneiva08, 2020-08-13:
@pointhi thanks for the tip
for linux mint started to work when installed with following command

`sudo apt-get install python-yaml`

chschlue, 2020-08-13:
So the interpreter doesn't barf on importing a non-existent module as long as it's not actually used? Interesting. TIL.

pointhi, 2020-08-14:
@chschlue yep, that's python. You can also import packages inside a function and so on.

calebreister, 2020-08-19:
Another potential option is to just use the built-in JSON module.

evanshultz, 2020-09-21:
@calebreister 
You've mentioned using JSON many times in many comments. While we have a lot to do with the GitLab migration and v6, but U haven't seen anybody comment. Personally, I am against it FWIW. The extra decoration makes it less approachable and readable than YAML, which I think it more important for the varied capabilities of users of our scripts. One simple dependency isn't too much to ask for the extra friendliness of YAML, IMO.

That being said, it's just my standpoint as of today. I don't mean to start a war or battle, but since this isn't the first time I've seen a similar comment I thought it was past time for somebody to respond to you about it.

calebreister, 2020-09-21:
@evanshultz Makes sense. My main issue with YAML is that it tends to be error-prone due to the significant whitespace. Since YAML is backwards-compatible with JSON, it works fine to feed JSON files into the YAML module (which is in fact what I did in my scripts). Personally, I find JSON more user-friendly by virtue of simplicity, but it is a user-by-user decision. Perhaps it would be a good idea to convert the global config files over to JSON so that it is at least possible for users to opt to use the built-in JSON module.