pointhi/kicad-footprint-generator #598: Footprint generator for AC/DC, DC/DC converters

Werni2A, 2020-08-30:
Hi, I was wondering if there exists any footprint generator for packages like the following. These packages are widely used in AC/DC and DC/DC converters therefore a single generator would be useful.

Hence I haven't found any script I will give it a try to write my own generator. 
- Do you know if there is something special I need to keep in mind?
- What is the preferred type of generator? It looks like the YAML based variants are state-of-the-art?

[image]

Franck78, 2020-09-01:
It looks like this script
https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/general/StandardBox.py

provides a function to call from your script that will do most of the job.

and.......

DC/DC AC/DC converter script already exists !!!!!!!!!

Werni2A, 2020-09-02:
@Franck78 Good that I asked. I will take a look at the script in a few days :+1: 

> DC/DC AC/DC converter script already exists !!!!!!!!!

Do you mean there is another script that does the conversion or the one you linked above? If there is another one, could you point me into the right direction?

Franck78, 2020-09-02:
When I look at the KiCad lib AC/DC DC/DC I see a lot of footprint like you RECOM.
So, I think it's just a single line (for each footprint you want) to add in the generator.py (or similar name) found in the AC/DC DC/DC subdir.

Werni2A, 2020-09-02:
I directly skipped the files in [Converter_DCDC](https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Converter_DCDC) as I read the function name "QFN" but they really seem to generate those kind of packages. Thanks.

closed 2020-09-02 by Werni2A