pointhi/kicad-footprint-generator #620: Reproducable builds, Basic CI

cpresser, 2020-09-10:
Hi,

while working on #604 I did some manual checking if the changes to KicadModTree break any existing footprints.
My approach was to run the bga and nolead generators with `find *.yaml -exec `. Then repeat the same on the master and compare results with diff.
Fortunately diff has a `-I` option so I could do `diff -ru -I tedit _604/ _m/ ` to ignore the timestamp lines.

When I was done my thought was 'why not automate that in a CI pipeline'. For that it would be nice if there is `--reproducible` command line argument to fix the timestamp.

Any comments? Good or bad idea? Implementation hints?

calebreister, 2020-09-10:
I like the idea of making scripts reproducible, but I'm wondering if the scripts directory is too much of a mess to make it work reliably. Also, I'm beginning to wonder if it would be a good idea to port the data files to JSON since support is built into the python standard libraries.

cpresser, 2020-09-10:
I would start with the IPC generators and perhaps add more later.
Regarding the JSON support, does it also support comments in the JSON file? That's a feature I wouldn't want to miss.

calebreister, 2020-09-10:
JSON doesn't support comments, though you could embed comments in the data and disregard it in the code. Just add a string to your object and name it "_" or something.